xsacha/Sachesi #56: Running Sachesi fails on Fedora 21 with libcrypto missing error

tlnagy, 2015-04-15:
I downloaded and extracted Sachesi on my machine, but when I ran the executable I got: 

```
./Sachesi: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or directory
```

I'm running Fedora 21. Any suggestions?

xsacha, 2015-04-16:
What version of libcrypto do you have? You would need to ensure the SSL versions are synced.

tlnagy, 2015-04-16:
Ah, I'm using version 1.0.1k:

```
[tnagy@localhost ~]$ sudo find / -name libcrypto*
/usr/lib64/libcrypto.so.10
/usr/lib64/libcrypto.so.1.0.1k
```

tlnagy, 2015-04-16:
Any recommendations on how to proceed?

xsacha, 2015-04-16:
I thought I was using a static lib to avoid this problem but it looks like you can either: install 1.0.0 as well (easiest) or recompile Sachesi (pretty easy if you have Qt Dev tools)

tlnagy, 2015-04-17:
I'm trying to compile after installing the `qt-devel` set in Fedora, I'm getting the following error

```
Project MESSAGE: Warning: unknown QT: quick
Project MESSAGE: Warning: unknown QT: qml
RCC: Error in 'translations.qrc': Cannot find file 'translations/English.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Russian.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/French.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Spanish.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/German.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Dutch.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Chinese.qm'
RCC: Warning: No resources in 'translations.qrc'.
```

tlnagy, 2015-04-17:
I forgot to include my qmake version:

```
[tnagy@localhost Sachesi]$ qmake-qt4 -v
QMake version 2.01a
Using Qt version 4.8.6 in /usr/lib64
```

tlnagy, 2015-04-17:
So I installed qt5 as follows:

```
sudo dnf install qt5-qtbase qt5-qtbase-devel qt5-qtdeclarative qt5-qtdeclarative-devel qt5-qtquickcontrols qt5-qtgraphicaleffects
```
Now, I'm just getting the missing translation file errors:

```
[tnagy@localhost Sachesi]$ qmake-qt5
RCC: Error in 'translations.qrc': Cannot find file 'translations/English.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Russian.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/French.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Spanish.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/German.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Dutch.qm'
RCC: Error in 'translations.qrc': Cannot find file 'translations/Chinese.qm'
RCC: Warning: No resources in 'translations.qrc'.
```

Looking at `translations/` the links to `.qm` should `.ts` instead, right?

thurask, 2015-04-17:
You have to compile translations into .qm files with

    $ lrelease

tlnagy, 2015-04-17:
Ah thanks! So I was finally able to get it to compile by running

```
[tnagy@localhost Sachesi]$ lrelease-qt5 translations/*
[tnagy@localhost Sachesi]$ sudo dnf install openssl-devel systemd-devel libusb-devel
[tnagy@localhost Sachesi]$ qmake-qt5 && make
```

closed 2015-04-17 by tlnagy