xsacha/Sachesi #60: Cant connect any more - Detected 1 Blackberry USB device(s) in Autodetect Mode.

JoinJoin, 2015-06-11:
All I get is "Detected 1 Blackberry USB device(s) in Autodetect Mode." and icon spinning. 
Last time I used Sachesi, I updated to 10.3.1.2576. 

Linux 3.19.0-18-generic x86_64 GNU/Linux, KDE 15.04

xsacha, 2015-06-11:
Is the issue only in 10.3.1 firmware? As in it worked in 10.3.0 and earlier?

stmblfrcr, 2015-06-11:
Maybe you need to do this https://mohskitchen.wordpress.com/2014/11/08/using-sachesi-with-debian/

bb010g, 2015-06-12:
Do you have ethernet enabled?

JoinJoin, 2015-06-12:
> Is the issue only in 10.3.1 firmware? As in it worked in 10.3.0 and earlier?

Yes, it worked with all the versions before 10.3.1.2576 because this is the only way I can update BB OS.  
Just to be sure, I compiled Sachesi from latest source code but this changed nothing. 

> Do you have ethernet enabled?

Yes, unless you mean something else by this :)

stmblfrcr, 2015-06-12:
Did you actually click the link above and/or tried ifconfig to see if you got an IP assigned?

JoinJoin, 2015-06-12:
> Did you actually click the link above and/or tried ifconfig to see if you got an IP assigned?

Yes I did. And yes, it has a IP address. 
It worked before and I have not made any changes to my network set-up.

JoinJoin, 2015-06-17:
Any news or ideas how to fix this? Sachesi is only way for me to update my BlackBerry because I am in a country where BB are not sold hence none of the telcos will push out updates.

xsacha, 2015-06-17:
Try reconnecting the USB to your Blackberry as you open Sachesi until it 'catches'.
Do you get a pop up on your phone asking whether you want to install drivers / charge / etc? If so, it thinks it hasn't connected to a computer.

JoinJoin, 2015-06-17:
> Try reconnecting the USB to your Blackberry as you open Sachesi until it 'catches'. 

Made no difference 

> Do you get a pop up on your phone asking whether you want to install drivers / charge / etc?

I get a pop-up, telling me: "Device connected to a computer" 

Sachesi has same spinning icon, "Detected 1 Blackberry USB device(s) in Autodetect Mode" message and "Talking to 2 possible device(s)."

In my terminal window, I can see:

```"Communication Error: 4 (Socket operation timed out) from 169.254.211.217" ```
and it gets repeated over and over. 

while ifconfig shows

```
           usb0      Link encap:Ethernet  HWaddr 4a:9d:XX:XX:XX:XX  
          inet addr:169.254.211.218  Bcast:169.254.211.219  Mask:255.255.255.252
          inet6 addr: XX::XXXX:XXXX:XXXX:74fe/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:65 errors:0 dropped:0 overruns:0 frame:0
          TX packets:227 errors:0 dropped:0 overruns:0 carrier:0
         collisions:0 txqueuelen:1000 
```

xsacha, 2015-06-17:
Thanks! That looks like something I can fix from the Sachesi side then. Nothing wrong with your end.
Are you using Sachesi 2.0.1 from github releases or did you compile it yourself? What is your OpenSSL version?

daradib, 2015-06-18:
I have the same problem. 10.3.0 worked but I used Sachesi to upgrade to software release 10.3.1.1865 and can no longer connect. I'm running Debian 8 with openssl [1.0.1k-3](http://metadata.ftp-master.debian.org/changelogs/main/o/openssl/stable_changelog). Same error messages, except slightly different IP address (as expected) and 1 possible device not 2 (disconnected from eth0). I've compiled 2.0.0rc and 312add4 (current master) and tried the 2.0.1 binary from GitHub releases.

JoinJoin, 2015-06-19:
I downloaded 201 and guess what nothing has changed :( 
I get exactly the same problem as described before.

xsacha, 2015-06-19:
Sorry :( It's hard to fix when I can't reproduce. Maybe if I install the
same distro I will get it.

daradib, 2015-06-20:
@xsacha What distro, OpenSSL version, libusb version, and BlackBerry release are you using, if you don't mind me asking?

I ran `strace` on both 2.0.0rc compiled from source and 2.0.1 from GitHub releases. Not sure what errors to expect. Here are some examples that seem to suggest USB device problems. I'll send you the full strace by email. Let me know if any more info would be helpful.

    readlink("/sys/devices/pci0000:00/0000:00:14.0/usb2", 0x7ffd8c75dbc0, 1024) = -1 EINVAL (Invalid argument)
    readlink("/sys/devices/pci0000:00/0000:00:02.0", 0x7ffd8c75cc00, 1024) = -1 EINVAL (Invalid argument)
    poll([{fd=5, events=POLLIN}, {fd=6, events=POLLIN}, {fd=12, events=POLLIN}, {fd=23, events=POLLIN}], 4, 0) = 1 ([{fd=5, revents=POLLIN}])
    poll([{fd=5, events=POLLIN}, {fd=6, events=POLLIN}, {fd=12, events=POLLIN}, {fd=23, events=POLLIN}], 4, 0) = 1 ([{fd=5, revents=POLLIN}])
    read(5, "\2\0\0\0\0\0\0\0", 16)         = 8
    poll([{fd=5, events=POLLIN}, {fd=6, events=POLLIN}, {fd=12, events=POLLIN}, {fd=23, events=POLLIN}], 4, 0) = 0 (Timeout)
    read(5, 0x7ffd8c75df50, 16)             = -1 EAGAIN (Resource temporarily unavailable)
    poll([{fd=5, events=POLLIN}, {fd=6, events=POLLIN}, {fd=12, events=POLLIN}, {fd=23, events=POLLIN}], 4, 0) = 0 (Timeout)
    futex(0x80de78, FUTEX_WAIT_PRIVATE, 2, NULL) = -1 EAGAIN (Resource temporarily unavailable)
    recvmsg(6, 0x7ffd8c75dda0, 0)           = -1 EAGAIN (Resource temporarily unavailable)
    read(5, 0x7ffd8c75df50, 16)             = -1 EAGAIN (Resource temporarily unavailable)

xsacha, 2015-06-20:
I'm using Ubuntu 14.10 and Ubuntu 15.04. Blackberry 10.3.1.1xxx

JoinJoin, 2015-06-20:
I have Ubuntu 15.04. Blackberry 10.3.1.1779  
PC sees the phone, phne sees the PC and even Sachesi detects the phone but is not connecting to it. 

Is it possible you add some temporary code to make it more verbose so I can post the log for you?

xsacha, 2015-06-20:
Sure. I might permanently add a 'Troubleshooting' button that gives a detailed log of what it is trying to do.

daradib, 2015-06-21:
Strange, I can't reproduce this today. Using same binaries and haven't restarted phone or laptop, only suspended.

I was connecting through a ThinkPad dock before instead of directly. Will test dock again on Monday and report back. It's possible my problem was different. @JoinJoin are you using the same USB port as you were when Sachesi worked for you?

daradib, 2015-06-21:
@JoinJoin Can you try inspecting packets with `tcpdump` or `tshark`? Use the `-i usb0` option, replacing `usb0` with the appropriate interface from `ifconfig`. The traffic is encrypted with SSL/TLS, but there might be some useful information in the handshake.

69goat, 2015-06-22:
I couldn't connect after updating to 15.04.   This is what I did.  Edit /etc/network/interfaces as root and add this:

auto usb0 
iface usb0 inet dhcp

Then save it and run
sudo /etc/init.d/networking restart

xsacha, 2015-06-22:
Did this work? Ubuntu is meant to automatically run dhcp on usb devices out-of-box.

69goat, 2015-06-22:
Yes.  I am using it now and have been for over a month with no connection problems.
I also had problems with my Arch based Manjaro.   I had to run sudo dhcpcd every time to get to connect.  I think I have that fixed now also, but still testing it.

daradib, 2015-06-22:
@JoinJoin and I had IP addresses assigned by the phone, so it may be a
different issue.

xsacha, 2015-06-22:
IP for usb0 is always assigned by computer and will be +1 of phone IP. You
NEED to always do dhcp, but Ubuntu should be doing that automatically.
Otherwise your phone is not discoverable.
Also the IP is generally random unless you are using developer mode, in
which case it is 169.254.0.1 or a whatever you chose.

JoinJoin, 2015-06-22:
> I couldn't connect after updating to 15.04. This is what I did. Edit /etc/network/interfaces as root and > add this:

> auto usb0
> iface usb0 inet dhcp

> Then save it and run
>  ''' 
sudo /etc/init.d/networking restart ```

69goat, 2015-06-23:
Good. I am glad it worked for you also.

hexs666, 2015-07-01:
haveing same issue as OP but i am Running Fedora 21; any one have any suggestions as I have checked ifconfig i am getting ip address. thanks for any insight

JoinJoin, 2015-07-04:
hexs666, did you follow 69goat instructions? If not, do it and you will be fine. 
Something has changed in Kubuntu default network setting and looks like this is the cause.

hexs666, 2015-07-04:
@JoinJoin I cannot follow the exact instructions because Fedora doesn't mean utilize the same config files, I have looked at config scripts and all are set for auto and DHCP; so at this point not sure what the to do.

69goat, 2015-07-04:
I have some suggestions, but from Arch use.  Sometimes just sudo dhcpcd works.  It is also best to boot the pc with the phone plugged in, I think dchpcd times out if there are no connections found.  

Run systemctl list-unit-files and see if dhcpcd.service and dhcpcd@.service are enabled.
If not, run systemctl enable dhcpcd.service and systemctl enable dhcpcd@XXXXX.service.
Run ls /sys/class/net or systemctl status dhcpcd -l  to see what the XXXXXX of the usb connection is.  Maybe usb0 or enp something. Then systemctl start dhcpcd@XXXXX.service

hexs666, 2015-07-07:
@69goat are you talking about enable-ling the dhcp server daemon....my laptop has never been a dhcp server and not sure why i need to have the packages and service installed since previously this was not needed.

69goat, 2015-07-08:
Like I said, that info is from Arch use.  Dhcpcd isn't installed in Lubuntu either.  Can you see what connection manager you use?

69goat, 2015-07-08:
Another suggestion I found here http://www.tuxradar.com/content/diagnose-and-fix-network-problems-yourself.  "On Fedora and Red Hat the filename would be of the form /etc/sysconfig/network-scripts/ifcfg-eth*, on SUSE it would be /etc/sysconfig/network/ifcfg-eth*, and on Ubuntu it would be /etc/network/interfaces."  Perhaps edit that file like I did with Ubuntu??

hexs666, 2015-07-08:
reviewed all the networking scripts under /etc/sysconfig/networking-scripts and there are none defined for enpXXXXX .
logs in messages as they pertain to the activation and auto ip assignment of the enpXXXX device as follows:

Jul  8 10:14:34 SDLCNU2102ZRG kernel: usb 1-1: new high-speed USB device number 6 using xhci_hcd
Jul  8 10:14:34 SDLCNU2102ZRG kernel: usb 1-1: New USB device found, idVendor=0fca, idProduct=8017
Jul  8 10:14:34 SDLCNU2102ZRG kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul  8 10:14:34 SDLCNU2102ZRG kernel: usb 1-1: Product: BlackBerry
Jul  8 10:14:34 SDLCNU2102ZRG kernel: usb 1-1: Manufacturer: Research In Motion, Ltd.
Jul  8 10:14:34 SDLCNU2102ZRG kernel: usb 1-1: SerialNumber: 463D01BEA3F60AED00D60639D2675F220E6B891C
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.0: MAC-Address: a6:e4:b8:04:30:b5
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.0 usb0: register 'cdc_ncm' at usb-0000:25:00.0-1, CDC NCM, a6:e4:b8:04:30:b5
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.2: MAC-Address: a6:e4:b8:04:30:ba
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.2 usb1: register 'cdc_ncm' at usb-0000:25:00.0-1, CDC NCM, a6:e4:b8:04:30:ba
Jul  8 10:14:34 SDLCNU2102ZRG mtp-probe: checking bus 1, device 6: "/sys/devices/pci0000:00/0000:00:1c.7/0000:25:00.0/usb1/1-1"
Jul  8 10:14:34 SDLCNU2102ZRG mtp-probe: bus: 1, device: 6 was not an MTP device
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.0 enp37s0u1: renamed from usb0
Jul  8 10:14:34 SDLCNU2102ZRG kernel: IPv6: ADDRCONF(NETDEV_UP): enp37s0u1: link is not ready
Jul  8 10:14:34 SDLCNU2102ZRG kernel: IPv6: ADDRCONF(NETDEV_UP): enp37s0u1: link is not ready
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.0 enp37s0u1: network connection: connected
Jul  8 10:14:34 SDLCNU2102ZRG kernel: IPv6: ADDRCONF(NETDEV_CHANGE): enp37s0u1: link becomes ready
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.0 enp37s0u1: kevent 12 may have been dropped
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.0 enp37s0u1: 480 mbit/s downlink 480 mbit/s uplink
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): carrier is OFF
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): new Ethernet device (driver: 'cdc_ncm' ifindex: 15)
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): exported as /org/freedesktop/NetworkManager/Devices/5
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): device state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): preparing device
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  ifcfg-rh: add connection in-memory (c7fb23a3-532d-4efa-9826-d62a0c642c6e,"Wired connection 1")
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): created default wired connection 'Wired connection 1'
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): link connected
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): device state change: unavailable -> disconnected (reason 'carrier-changed') [20 30 40]
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  Auto-activating connection 'Wired connection 1'.
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: starting connection 'Wired connection 1'
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 1 of 5 (Device Prepare) scheduled...
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 1 of 5 (Device Prepare) started...
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): device state change: disconnected -> prepare (reason 'none') [30 40 0]
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 2 of 5 (Device Configure) scheduled...
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 1 of 5 (Device Prepare) complete.
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 2 of 5 (Device Configure) starting...
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): device state change: prepare -> config (reason 'none') [40 50 0]
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 2 of 5 (Device Configure) successful.
Jul  8 10:14:34 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 2 of 5 (Device Configure) complete.
Jul  8 10:14:34 SDLCNU2102ZRG kernel: cdc_ncm 1-1:1.2 enp37s0u1i2: renamed from usb1
Jul  8 10:14:35 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1i2): carrier is OFF
Jul  8 10:14:35 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1i2): new Ethernet device (driver: 'cdc_ncm' ifindex: 16)
Jul  8 10:14:35 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1i2): exported as /org/freedesktop/NetworkManager/Devices/6
Jul  8 10:14:35 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1i2): device state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Jul  8 10:14:35 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1i2): preparing device
Jul  8 10:14:35 SDLCNU2102ZRG kernel: IPv6: ADDRCONF(NETDEV_UP): enp37s0u1i2: link is not ready
Jul  8 10:14:35 SDLCNU2102ZRG kernel: IPv6: ADDRCONF(NETDEV_UP): enp37s0u1i2: link is not ready
[... cut ...]
Jul  8 10:14:36 SDLCNU2102ZRG NetworkManager[12888]: <info>  (enp37s0u1): Activation: Stage 5 of 5 (IPv4 Commit) complete.
Jul  8 10:14:36 SDLCNU2102ZRG avahi-daemon[946]: Joining mDNS multicast group on interface enp37s0u1.IPv4 with address 169.254.61.34.

From what i can tell the DHCP client request is being honored by the BBOS10 and the interface is accepting and assignin the appropiate IP; but i am unable to ping my gateway (BB10 device) with the -1 IP address ; i have also flushed uiptables for the session to rule out firewall and have set SElinux to permissive

69goat, 2015-07-10:
When I was getting no connection with Arch I had similiar info from journalctl -f.  I think I have that fixed by creating a file to /etc/systemd/network/wired.network.  All it says is this.
[Match]
Name=en*

[Network]
DHCP=yes

Check the settings/about/network usb IPv4 when plugged in and try to just ping that ip.  Even when I couldn't connect to Sachesi I could still  ping the phone.

f2re, 2015-09-07:
Hi! i'am trying to connect BB on mac OS and have similar error. in macos ifconfig -a dont have **usb0** device, but have **en6** device:

en6: flags=863<UP,BROADCAST,SMART,RUNNING,SIMPLEX> mtu 1500
	ether 42:6f:2a:77:c3:9e
	inet 169.254.0.2 netmask 0xfffffffc broadcast 169.254.0.3
	media: autoselect (1000baseT <full-duplex>)
	status: active

any ideas about this? i trying all versions - it is not connectiong to device; OS 10.3.1.1779

xsacha, 2015-09-07:
That looks fine. It should be able to connect to 169.254.0.2.
Can you see the log in Sachesi? There'll be a button you can press to view it.

f2re, 2015-09-08:
no, button can't press. button is active (not disabled) but when ckicked - nothing happend.

xsacha, 2015-09-08:
Thanks. Fixed in latest commit!

closed 2015-09-08 by xsacha

prasadeng, 2017-03-10:
I'm using Linux mint 17 and schesi 2.2 latest version once I connected and back up and update also worked but now it s spinning not connect to bb. can u fix this

xsacha, 2017-03-10:
Not sure what went wrong there but if it finds the network interface it will just work

prasadeng, 2017-03-10:
thanks i connect network interface still same mmmmmmmmmmmmmmmmmmm...........

xsacha, 2017-03-10:
Can you run: ifconfig
In terminal?
See IP for blackberry interface.

prasadeng, 2017-03-10:
this is screen picture Ethernet port didn't get addresses.

prasadeng, 2017-03-10:
[image]
this is screen print didn't got ip address.

xsacha, 2017-03-10:
Yeah your Blackberry hasn't connected as an ethernet device so you can't talk to it.

xsacha, 2017-03-10:
Which device is it? Have you enabled developer mode on the phonr?

prasadeng, 2017-03-10:
Q10 yes I enable developer mode

prasadeng, 2017-03-10:
osversion 10.3.3.746 sqn 100-3

xsacha, 2017-03-10:
Can you provide output of lsusb -v?

It is connected via USB right?

prasadeng, 2017-03-10:
lsusb (usbutils) 007

prasadeng, 2017-03-10:
Bus 002 Device 005: ID 0fca:8014 Research In Motion, Ltd. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0fca Research In Motion, Ltd.
  idProduct          0x8014 
  bcdDevice            2.40
  iManufacturer           1 Research In Motion, Ltd.
  iProduct                2 BlackBerry Mass Storage Device
  iSerial                 3 8DAA01F45E6E2388D3592F243A4339851C18E5AA
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
[... 786 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval             255
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             2
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0103 power enable connect
   Port 2: 0000.0100 power
Device Status:     0x0001
  Self Powered

xsacha, 2017-03-10:
Yeah your phone is in mass storage mode. It needs the other mode.

bInterfaceClass 8 Mass Storage
bInterfaceSubClass 6 SCSI
bInterfaceProtocol 80 Bulk-Only

klango, 2017-11-18:
Hi I ended up here when had the problem connecting my Passport (10.3.3.1435)

1. Sachesi 2.0.3
2. Dev mode on
3. Mass storage off
4. Errors
 "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" and
"Communication Error: 203 (Error downloading http://192.168.2.105/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.2.105"

Tried all suggested procedures with no luck. I changed the IP Adress on BB10 under Development mode to 169.254.0.5 (instead default 169.254.0.1)
Sachesi then was able to recognize. I hope this can help someone.

Cheers