xsacha/Sachesi #66: System Error - MSVCP.dll is missing

williamrjribeiro, 2015-07-12:
I get the following System Error message on a popup when I try to run Sachesi 2.0.2 on a fresh install of Windows 10 Pro Insider Preview Build 10162:
"The program can't start because MSVCP120.dll is missing from your computer. Try reinstalling the program to fix this problem".

Can anyone check if it also happens on another version of Windows?

xsacha, 2015-07-12:
Those errors will happen on all Windows. It is saying msvc 2013 redistributable x86 isn't installed.
I guess I missed a DLL.
I will correct in next version, thanks! You are first one to tell me :p

williamrjribeiro, 2015-07-12:
Hum, I see the " msvcr120.dll" file next to "Sachesi.exe" and I have installed the "vcredist_x64" but it still doesn't work. Strange.

xsacha, 2015-07-12:
Yeah it needs to be x86 version because the binary is 32-bit. Not related
to your system being x64.
Looks like there is two DLLs I needed to include (msvcr and msvcp).

closed 2015-09-06 by xsacha