xsacha/Sachesi #68: Crashes on Start

DigiAngel, 2015-07-13:
Windows 7 64 Bit:
~~~
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	Sachesi.exe
  Application Version:	2.0.1.0
  Application Timestamp:	558d1ec2
  Fault Module Name:	ig4icd32.dll
  Fault Module Version:	8.14.10.1930
  Fault Module Timestamp:	4aba6fc2
  Exception Code:	c0000005
  Exception Offset:	002095ed
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033
  Additional Information 1:	0a9e
  Additional Information 2:	0a9e372d3b4ad19135b953a78882e789
  Additional Information 3:	0a9e
  Additional Information 4:	0a9e372d3b4ad19135b953a78882e789
~~~

What other information can I provide?  Thank you.

xsacha, 2015-07-13:
Looks like graphics driver (Intel) issue.
Can you try the newer version, 2.0.2? It has a newer Qt with dynamic graphics backend.

DigiAngel, 2015-07-14:
Thank you....surprisingly, that actually is 2.0.2.  The title bar says 2.0.2, but the file output above says 2.0.1.0...which is weird.  This is from the 2.0.2 7zip file, which I had just downloaded.  This is running on a Mac Mini with a Mobil Intel 945 Express chipset.  Thank you.

xsacha, 2015-09-06:
Sorry for not getting back to this issue. Do older versions work? For example, 1.4?
I think the newer Qt's may have deprecated your GPU :(

DigiAngel, 2015-09-06:
Thanks Sacha...you can kill this...I went to a different machine and this worked fine thank you.

closed 2015-09-06 by DigiAngel