xsacha/Sachesi #69: Not connecting from linux to ОС 10.3.2.798

mcwolfmm, 2015-07-22:
Sachesi v.2.0.2 not connecting to q10 with OS 10.3.2.798.
"Detected 1 Blackberry USB device(s) in Autodetect mode" and nothing more.

In this OS no way to select "USB access mode" between windows and mac.

mariusvam, 2015-07-25:
I have the same issue on kde ultimate distro 64 bit. and I can do anything now because my phone is blinked and I have no windows box to run the autoloader or other erasing utilities. I will try to install windows 7 in a virtual disk.

Is the any other version that may work in kubuntus kde?

mcwolfmm, 2015-07-25:
As it turned out, the problem is not sachesi. The problem is the network manager.

I decided as I turned the phone in usb port and waited for the network manager to detect it. Then I run (network manager) and forbid / remove interface (usb0). Then run the terminal and configure manually (sudo dhclient usb0). Then start sachesi and have no problem with the connection.

closed 2015-09-06 by xsacha

klango, 2017-11-18:
Hi I ended up here when had the problem connecting my Passport (10.3.3.1435)

1. Sachesi 2.0.3
2. Dev mode on
3. Mass storage off
4. Errors
 "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" and
"Communication Error: 203 (Error downloading http://192.168.2.105/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.2.105"

Tried all suggested procedures with no luck. I changed the IP Adress on BB10 under Development mode to 169.254.0.5 (instead default 169.254.0.1)
Sachesi then was able to recognize. I hope this can help someone.

Cheers

paoletto, 2017-11-25:
last post before closure had the solution for me. Problem was indeed the bloody network manager.