xsacha/Sachesi #73: Downloading files from BB servers indicates over 100%

polytan02, 2015-08-13:
A picture is always better than a long text : 

[image]

polytan02, 2015-08-13:
Actually, the downloading was blocked on this file in an infinite loop.

I unselected it from the list to be able to continue retrieving files. Strange.

xsacha, 2015-08-13:
They blocked the carrier-specific 'bell' file? That is interesting!
Can you tell me what the MCC/MNC you used was and which firmware you were
downloading so I can replicate?
I will have a solution for blocked files then.

polytan02, 2015-08-13:
MCC : 302 (Canada)
MNC : 640 (Bell)

I was on 10.3.2.2204 and I did a upgrade (non destructuve) via your app to 10.3.2.2339.

Great app by the way ;)

xsacha, 2015-09-08:
I think possibly fixed in latest commit. I didn't get exactly the same issue as you but similar.

closed 2015-09-08 by xsacha