xsacha/Sachesi #80: Windows package missing msvcp120.dll file

lordong, 2015-10-11:
Please add msvcp120.dll into the windows zip package next time.

I download msvcp120.dll from http://cn.dll-files.com/msvcp120.dll.html (32bit), it is work.

Thanks.

xsacha, 2015-10-11:
Thanks, I will remember that.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        From: lordongSent: Sunday, 11 October 2015 10:44 AMTo: xsacha/SachesiReply To: xsacha/SachesiSubject: [Sachesi] Windows package missing msvcp120.dll file (#80)Please add msvcp120.dll into the windows zip package next time.

I download msvcp120.dll from http://cn.dll-files.com/msvcp120.dll.html (32bit), it is work.

Thanks.

jbuc89, 2016-01-20:
Or just download the x86 version of Visual C++ Redistributable Packages for Visual Studio 2013.

https://www.microsoft.com/en-us/download/details.aspx?id=40784

arynadam, 2017-02-19:
You can also download the 64bit version from http://www.dll4free.com/msvcp120.dll.html
32bit worked for me

xsacha, 2018-05-12:
This was included in some versions but not others.
Sorry for that. If you compile yourself you will not have such an issue.

closed 2018-05-12 by xsacha