xsacha/Sachesi #81: Windows version

DLewis13, 2015-10-14:
Still looking for Windows version of 2.03.

Thanks for the great work xsacha,
Dick Lewis

xsacha, 2015-10-14:
Was just waiting on a new Qt release. Should be able to push one out now.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        From: DLewis13Sent: Thursday, 15 October 2015 12:51 AMTo: xsacha/SachesiReply To: xsacha/SachesiSubject: [Sachesi] Windows version (#81)Still looking for Windows version of 2.03.

Thanks for the great work xsacha,
Dick Lewis

Calvin413, 2015-12-01:
I am using Windows 10. My A/V keeps quaranteinig‎ one of the qt files. I have backed down to version 2.0.1 and it runs fine. I still have not seen a new version for Windows that I can run successfully.                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: ModeenFSent: Tuesday, December 1, 2015 14:09To: xsacha/SachesiReply To: xsacha/SachesiSubject: Re: [Sachesi] Windows version (#81)Any news?

xsacha, 2018-05-12:
Windows version was released.

closed 2018-05-12 by xsacha