xsacha/Sachesi #83: Error 6 (SSL handshake failed) 2.0.3

jtet, 2015-12-08:
Using the precompiled Sachesi for Mac Os X on El Capitan (10.11) the search tab of Sachesi reports "Error 6 (SSL handshake failed)". Search does not function as a result.
[image]

xsacha, 2015-12-09:
Sounds like different OpenSSL versions. I thought I compiled it as static.
What OpenSSL version is on your OSX El Capitan?

jtet, 2015-12-09:
```
~ openssl version
OpenSSL 0.9.8zg 14 July 2015
```

franzos, 2016-12-28:
I'm having the same issue.

OpenSSL 0.9.8zh 14 Jan 2016
MacOS 10.12.2

niko-lay, 2017-01-12:
Same issue on Siera 

``` console log
$ ./Sachesi
qt.network.ssl: QSslSocket: cannot resolve SSL_set_psk_client_callback
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_server_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSL_select_next_proto
qt.network.ssl: QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb
qt.network.ssl: QSslSocket: cannot resolve SSL_get0_next_proto_negotiated
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get0_next_proto_negotiated
qrc:/qml/generic/Installer.qml:90:25: Unable to assign [undefined] to QUrl
^C

$ openssl version
OpenSSL 1.0.2j  26 Sep 2016
```

xsacha, 2017-01-12:
Are you able to compile Sachesi from source? If someone with Sierra can compile it, I can distribute it here.

franzos, 2017-01-12:
I'll give it a shot - Downloading QT.

adin234, 2017-02-07:
any updates re sierra?

L1cardo, 2019-06-10:
@franzos   same issue.
`'openssl/rsa.h' file not found `

xsacha, 2019-06-10:
This is a different issue (runtime) whereas yours is build time.

pseuudonym404, 2021-01-25:
I realise this is an old one, but I'm also getting Error 6 (SSL handshake failed) on Kubuntu 20.10 with OpenSSL 1.1.1f  31 Mar 2020

I can probably help fix it, but I haven't tracked down which bit of code I need to work on yet, tips would be helpful. :)

xsacha, 2021-01-25:
You just need to recompile it with the newer OpenSSL that you are now using

pseuudonym404, 2021-01-25:
If that had worked, I wouldn't have posted a comment. I'd already built it myself...