xsacha/Sachesi #85: 0xc000007b

DLewis13, 2015-12-25:
2.0.3 failed to start, error 0xc000007b.
Ran sfc /scannow in Command Prompt, nothing to fix.
Searches do not help.
Running Window 7 SP1 Ultimate.

Thanks, Dick Lewis

DLewis13, 2016-01-18:
Looks like you need vc_redist.x86.exe, I had x64 installed.
After installing x86 version, 2.0.3 works fine.
Link to Microsoft site:

https://www.microsoft.com/en-us/download/details.aspx?id=48145

jbuc89, 2016-01-20:
The link above was for Visual C++ Redistributable for Visual Studio 2015 and did not work on my Windows 7 installation.

Here is the link for Visual C++ Redistributable Packages for Visual Studio 2013 which did work for Windows 7 and also go rid of the "MSVCR120.dll missing error message":

https://www.microsoft.com/en-us/download/details.aspx?id=40784

DLewis13, 2016-01-22:
Worked on mine, I already had the other versions installed. Including 2008, 2010, 2012 and 2013.
I only posted my final step, everyone's configuration is different.

christinegrant98, 2017-01-24:
just google the issue and found this guide to fix Error Code 0xc000007b worked for me as i have faced the same problem.
[http://errorcodespro.com/fix-error-code-0xc000007b-windows-7-10/](http://errorcodespro.com/fix-error-code-0xc000007b-windows-7-10/)

xsacha, 2018-05-12:
Duplicate of #62

closed 2018-05-12 by xsacha