xsacha/Sachesi #86: Add Support for Blackberry Priv

rheza, 2016-01-29:
Hi Guys, 

Is there any way to support Blackberry Priv ?
I look at the source code, its pretty much the same requirement to get the autoloder.

I don't have time to hack around this app. hopefully someone can make it. 

Thanks

xsacha, 2016-01-29:
Hi,
I dont have a priv but I assumed it is totally different.

gstanden, 2016-01-29:
It's android

"If BlackBerrys were as plentiful as Apples, upon compulsions, no man seven reasons would I give, for BlackBerry 10 reasons hath one to give, I."

Gilbert Standen after William Shakespeare, Henry IV‎

Sent from my BlackBerry Z30 OS10 smartphone on the Verizon Wireless 4G LTE network.
From: xsacha
Sent: Friday, January 29, 2016 10:40 AM
To: xsacha/Sachesi
Reply To: xsacha/Sachesi
Subject: Re: [Sachesi] Add Support for Blackberry Priv (#86)

Hi,
I dont have a priv but I assumed it is totally different.

---
https://github.com/xsacha/Sachesi/issues/86#issuecomment-176854193

Feakster, 2016-01-29:
You don't need Sachesi for the Priv because it runs Android, not BB OS 10.x. I you want to disable/uninstall most Android applications it's relatively simple. If you want to uninstall things like the default Google apps, you'll need to find some way of rooting your Priv.

thurask, 2016-01-30:
Do you want an Android app that does the usual BB10 stuff (app control, bar download, software lookup, &c), or an Android app that does the Android equivalent?

The second is impossible, and the first is (IMO) pointless.

gstanden, 2016-01-30:
HI Sacha,

It's not me who posted this question.

I only love true BlackBerry 10 devices. I looked at PRIV at the AT&T store and compared to bb10 it's primitive.

Cheers,

"If BlackBerrys were as plentiful as Apples, upon compulsions, no man seven reasons would I give, for BlackBerry 10 reasons hath one to give, I."

Gilbert Standen after William Shakespeare, Henry IV‎

Sent from my BlackBerry Z30 OS10 smartphone on the Verizon Wireless 4G LTE network.
From: thurask
Sent: Saturday, January 30, 2016 12:05 AM
To: xsacha/Sachesi
Reply To: xsacha/Sachesi
Cc: Gilbert Standen
Subject: Re: [Sachesi] Add Support for Blackberry Priv (#86)

Do you want an Android app that does the usual BB10 stuff (app control, bar download, software lookup, &c), or an Android app that does the Android equivalent?

The second is impossible, and the first is (IMO) pointless.

---
https://github.com/xsacha/Sachesi/issues/86#issuecomment-177081362

rheza, 2016-02-01:
i don't think it's different. the autoloader leaked for the priv named like this.

"bbry_qc8992_autoloader_user-na-none-AAC826.zip"

the priv use the same name scheme for the autoloader.

after extracted, its has two folder. one folder for fastboot to flash the file.
the second folder called "img", which all the file to be flashed when running the flash-all.bat on folder root.

The image,
[image]

I only need to get direct link for the all available autoloader for the priv.
What trick do you use to get the browser listed all available file for the bb10 firmware ?

Thanks

xsacha, 2017-05-13:
I make calls to the firmware servers that you can see in src/search source code.

closed 2017-05-13 by xsacha