xsacha/Sachesi #87: error 99 ssl

Corsari, 2016-02-06:
Hello

I'm trying out sachesi 2.0.3

Under windows 10 x64 I had a first issue, I had to install the visual studio x32 runtime since x64 was useless for the need of fixing a missing vc140 dll

Now it runs but as soon I lookup in the search tab (I've entered correct country and carrier code and infact he vodafone logo appears) at bottom of the window it appears "error 99 SSL context creation error" (sorry, I translate it from Italian) and the "Searching" button stays grayed forever

Can you kindly hint about this? The program is run as administrator

Thank you
R.

xsacha, 2016-02-06:
Oh I mustn't have included the right SSL files :( you can use the previous version 2.02 instead.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        From: CorsariSent: Saturday, 6 February 2016 5:01 PMTo: xsacha/SachesiReply To: xsacha/SachesiSubject: [Sachesi] error 99 ssl (#87)Hello

I'm trying out sachesi 203

Under windows 10 x64 I had a first issue, I had to install the visual studio x32 runtime since x64 was useless for the need of fixing a missing vc140 dll

Now it runs but as soon I lookup in the search tab (I've entered correct country and carrier code and infact he vodafone logo appears) at bottom of the window it appears "error 99 SSL context creation error" (sorry, I translate it from Italian) and the "Searching" button stays grayed forever

Can you kindly hint about this? The program is run as administrator

Thank you
R

Corsari, 2016-02-06:
Hello I went on with 199
it has downloaded the upgrade files
it started the upgrade process but looks like it doesn't calculate the available space in relation with the selected files
while upgrading it stopped with error of not enough available space :-(
I had to delete some few megabytes to complete the last two files of the upgrade

Well...one question.. wasn't this process to be supposed to overwrite the current files and so avoid the issue of "not enough space" got from the OTA upgrade?

Thank you for this info.

xsacha, 2016-05-16:
Yes, sorry, that issue was fixed in the later versions.
Sachesi doesn't handle the internal upgrade. Blackberry keeps a backup of the last state in case the upgrade fails. This is why it runs out of space.

Not sure why the SSL error in 2.03 as no one else has reported. I did change build environment only for 2.03. Probably 2.02 is fine for you?

voidelixir, 2016-12-11:
I just tried 2.03 and have the same Issue. Also, 2.03 works just fine. The "error 99" I mean.

xsacha, 2018-05-12:
Looks subtly related to #83 (but Windows version).
Most people don't have issues but there is a few that do.

xsacha, 2018-05-20:
The latest version should fix your issue

closed 2018-05-20 by xsacha