xsacha/Sachesi #89: Help please. Cannot get Sachesi to run on anything

DWRThumper, 2016-05-15:
I have tried several version on Windows10, Windows 8, Windows XP, Ubuntu 14.04, and XUBuntu. All Windows machines have either a DLL error or (on XP) a message that it is not a valid ...32 application. The Linux based boxes do nothing. I click on the icon and nothing happens.

I am not very well versed in Ubuntu, but if someone could give me step by step (Instructions for a dummy, please) instructions, I might be able to get it installed. 

I am in Blackberry Q10 10.3 Blackberry ID hades and am getting no love from Blackberry or Link. Sachesi seems to be my only hope, but as I say, I cannot get it to run.

xsacha, 2016-05-15:
Hi.
XP support was dropped. What is the DLL error on Windows? That will be easy to remedy.

DWRThumper, 2016-05-15:
Booting that box up now. So this goes easily, here is what I did. I grabbed the 2.03 and 2.02 versions and expanded them on the Windows (8 and 10) boxes. Do I need to grab the associated zip file as well?

Thank you for the fast response! Awesome!! :)

DWRThumper, 2016-05-15:
Great... Winders has to do updates... Can you assist me with the Ubuntu 14.04 install?

xsacha, 2016-05-15:
There are multiple files and it requires extracting the zip prior to running so that it can locate these files.
If on a Linux box, run the app from a terminal so you can view any errors.

DWRThumper, 2016-05-15:
MSVCP120.dll is missing. This is the error on the Windows 8 box for version 2.0.2

VCRUNTIME140.dll is missing. This is the error on the Windows 8 box for version 2.0.3

xsacha, 2016-05-15:
MSVCP120 error: You need to install visual c++ redistributable for VS 2013 x86

DWRThumper, 2016-05-15:
OK, 2.0.2 is looking for a usb device and 'talking' to it. There is a password field. Is this for later use? It has been searching for some time now. How can I get Sachesi to recognize the Blackberry Q10 with 10.3.2.2836? I have had one of 'those kinds of weeks'. :)

xsacha, 2016-05-15:
The password is wants is your developer mode password. It needs that to connect to the device to perform backups and install apps.
Everything else won't need a password.

DWRThumper, 2016-05-15:
Should I put the phone in Developer mode to connect?

xsacha, 2016-05-15:
You can put it in development mode while Sachesi is running. No need to close it or anything.

DWRThumper, 2016-05-16:
So here is the big question, and I hope Sanchesi can do this. I want to back up the phone, then need to do a security wipe so I can get that stupid BBID to sync up. If I backup the phone, do the security wipe, then restore from the backup, will it overwrite the 'new' BBID' ?

xsacha, 2016-05-16:
You can only restore a backup to a device logged in to the same BBID as the backup was made from.
That is, backups are encrypted with the BBID used to create them.
So, in your situation, are there two different BBIDs?

DWRThumper, 2016-05-16:
Here is the problem. I upgraded to the 'newest' OS and then started getting BBID emails, asking me to confirm the email. I never got an email about BBID. I could not reset my password because I had no idea what the answer to the useless hint was. I searched Blackberry support and found a thread saying to delete the BBID and then create a new one. No one ever said I had to do a security wipe following that. So now the screenname is xxx-##### on the device, and the new B BID set up via the web is xxx-#####-#####. It will not let me set it to the one on my phone, stating that username already exists. The weird thing is, the BBID on the phone will not accept my password as it does on the web. I put it in, hit 'Verify' and nothing happens. If I put in garbage for a password, it immediately tells me it is invalid. If I hit 'forgot password', it shows me my new hint. Hmmm.... Sure wish i could simply make the screen name on my phone's BBID match the BBID on the web.

xsacha, 2016-05-16:
If you make a new BBID, there's no way to restore the backup from an old BBID. They probably should have mentioned that.
As for the password issue, you might be able to call up Blackberry to resolve that.

DWRThumper, 2016-05-16:
BB was no help at all. With your app, can I cherry pick files (texts, Remember, etc... stuff that cannot be copied via Windows Explorer) an save them to later push back into the phone?

xsacha, 2016-05-16:
It will save the data as a .bbb and can only be pushed back in to the phone that has logged in to the same BBID. Changing BBID will wipe the data.
It doesn't handle raw data since they encrypted backups.

DWRThumper, 2016-05-16:
So I guess I am screwed. I never confirmed the BBID on my phone, so am I totally screwed? Is there no way to get the sys. files via any tool? I only want the texts and 'Remember' files.

xsacha, 2016-05-16:
You can only grab those via an app installed on the phone that has access to them. There are some 'backup' apps on the app world.

closed 2018-05-12 by xsacha