xsacha/Sachesi #90: Can't  connect to BB10

mencacci, 2016-06-09:
Hello there, I compiled and installed on Debian Testing, but it doesn't connect to BB,,,
I disabled wi-fi, device in Develop Mode, enetred pw, but I get only:
"Communication Error: 4 (Socket operation timed out) from 169.254.0.1"

this is my route output
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         fritz.box       0.0.0.0         UG    600    0        0 wlp5s0
link-local      0.0.0.0         255.255.255.252 U     100    0        0 enp0s26u1u1
BLACKBERRY-B92C debian.local    255.255.255.255 UGH   100    0        0 enp0s26u1u1
192.168.178.0   0.0.0.0         255.255.255.0   U     600    0        0 wlp5s0
thanx in advance for your help

xsacha, 2016-06-09:
enp0s26u1u1 is the network device. Does it have an IP address from dhcpcd?
If not, run dhcpcd on it.
Otherwise, try rebooting your phone when you get this error.

mencacci, 2016-06-10:
Hello xsacha, here my ifconfig output:
enp0s26u1u1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.0.2  netmask 255.255.255.252  broadcast 169.254.0.3
        inet6 fe80::7b53:e814:7eab:206d  prefixlen 64  scopeid 0x20<link>
        ether 42:6f:2a:95:b6:71  txqueuelen 1000  (Ethernet)
        RX packets 68  bytes 25204 (24.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 219  bytes 20878 (20.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
route output:
Kernel IP routing table
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
link-local      0.0.0.0         255.255.255.252 U     100    0        0 enp0s26u1u1
BLACKBERRY-B92C debian.local    255.255.255.255 UGH   100    0        0 enp0s26u1u1

here there is wi-fi, but I tried without.
I tried  rebooting BB, but nothing....it detect BB on USB but always same error....
"Communication Error: 4 (Socket operation timed out) from 169.254.0.1"
any hints? thanx

mencacci, 2016-07-01:
Hello xsacha, I found that I NEED to be connected  by USB, and  to have Wi-Fi active, so it works fine.

xsacha, 2016-07-01:
You need USB because Blackberry doesn't allow any upgrades over wifi. Only app install.Wifi isn't needed at all.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        From: mencacciSent: Saturday, 2 July 2016 2:18 AMTo: xsacha/SachesiReply To: xsacha/SachesiCc: xsacha; CommentSubject: Re: [xsacha/Sachesi] Can't connect to BB10 (#90)Hello xsacha, I found that I NEED to be connected  by USB, and  to have Wi-Fi active, so it works fine.

dustybb, 2016-09-08:
Resolved on MacOSX Mac OS 10.11
*Error Message: Cannot communicate with connected device

1. Within the menu bar, select the Apple icon
2. Select System Preferences
3. Tap Networks
4. Note the name of the BlackBerry interface listed (Blackberry (en5 or 6))
5. Select the Plus sign to add a new interface
6. Within the Interface drop down, select the OTHER BlackBerry interface listed *So if it says en5 then you select en6*
7. Select Create
8. Select Apply
9. Restart the computer (unplug usb cable when doing restart) For me worked even without restart! 
 
But before you restart computer, delete the other blackberry interface that was listed, the one before you created a new one. So that there is no confusion between the two and it should work.

klango, 2017-11-18:
Hi I ended up here when had the problem connecting my Passport (10.3.3.1435)

1. Sachesi 2.0.3
2. Dev mode on
3. Mass storage off
4. Errors
 "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" and
"Communication Error: 203 (Error downloading http://192.168.2.105/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.2.105"

Tried all suggested procedures with no luck. I changed the IP Adress on BB10 under Development mode to 169.254.0.5 (instead default 169.254.0.1)
Sachesi then was able to recognize. I hope this can help someone.

Cheers

klango, 2017-11-19:
Just  a little update on this, I figured it out the connection problem.

While changing the IP Adress on BB10 under Development mode OR RIGHT AFTER connecting to the pc, you need to assign dhcpclient (dhcpclient <interface> ) in my case enp0**

closed 2018-05-12 by xsacha