xsacha/Sachesi #91: Translation source files are gitignored

cookiengineer, 2016-09-07:
See [gitignore#L26](https://github.com/xsacha/Sachesi/blob/master/.gitignore#L26) file.

The problem with the build setup is that the raw files (qm is the source format) are not tracked in git.
Therefore all the `RCC: Error in 'translations.qrc': Cannot find file '/translations/English.qm'` errors are a result of that.

Please push the source files (`translations/*.qm`) also to the repository, so that we can compile Sachesi.

Thanks much.

xsacha, 2016-09-07:
Hi,
In Qt, qm files are generated from the rd files. Generated files shouldn't be included in the source control. Qt should be running this prior to build.
Tested here, it builds from clean, which is how all platform released are built.

cookiengineer, 2016-09-07:
Okay, then I interpreted it correctly. As already merged, this issue is fixed then :)

closed 2016-09-07 by cookiengineer