xsacha/Sachesi #93: (Linux) Communication Error, even with udev rule

cookiengineer, 2016-09-07:
What should I do on a GNU/Linux system when there's a `Communication Error: 1`?

It should have proper USB rights, even with root it's not accessing the USB device properly.
Is there a verbose flag?

The device pops up properly with the `gnome-network-manager`, so I can use it via Tethering, use it as media storage device etc. pp.

```bash
cat /etc/udev/rules.d/51-blackberry.rules
SUBSYSTEM="usb", ATTR{idVendor}=="0fca", MODE="0666", GROUP="plugdev"
```

How can I debug this connection problem, it seems to be Sachesi specific, as everything else works perfectly.

cookiengineer, 2016-09-07:
ifconfig shows me this (USB connection):

```bash
enx96ebcdb8ae8a Link encap:Ethernet  HWaddr 96:eb:cd:b8:ae:8a  
          inet addr:169.254.0.2  Bcast:169.254.0.3  Mask:255.255.255.252
          inet6 addr: fe80::94eb:cdff:feb8:ae8a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:63 errors:0 dropped:0 overruns:0 frame:0
          TX packets:211 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:23998 (23.9 KB)  TX bytes:27983 (27.9 KB)
```

Why is Sachesi giving me this Error?

```bash
"Communication Error: 4 (Socket operation timed out) from 169.254.236.25"
```

This is the completely false B/C subnet. The IP of the device is the 169.254.0.1 and not 169.254.236.25.

Where can I debug more into that? Is this statically configured somewhere? If so, this must be documented in the README file.

xsacha, 2016-09-07:
Qt works out the IP devices and then Sachesi picks devices that match BlackBerry profile and attempts to communicate with them.
I am not sure why you see this error. Your environment sounds like it is set up correctly.

cookiengineer, 2016-09-07:
Is there something like a verbose compile flag or similar so that I can nail down the errors more easily?

Following some other tutorials to gain access to the device nmap can't find anything on the device, so that might be the issue, too. No service is available, also not in developer mode.

Any hints on what settings need to be checked to guarantee connections with the BB10 device?

xsacha, 2016-09-07:
The error would be from Qt or OpenSSL. If you compile them with debug it should provide more info. I'm not sure anything in Sachesi could explain it.

xsacha, 2016-09-07:
It just needs to appear as a network device on your computer. Your output in ifconfig looks correct.

xsacha, 2016-09-13:
After far-reaching your comment, it sounds like this error appears only 1 in 10 attempts and it is otherwise working?

You may have some other device/app causing this phantom IP to appear. For example virtualisation tools use addresses in that range. It is being picked up as the same config as a BlackBerry phone so it tests the IP. It will test all IPs though so this should not affect a connection.

cookiengineer, 2016-09-13:
Nope, I couldn't trace down the issue :(

Still can't connect to my Q5 and I'm kind of out of ideas to go after. Already in the state to try out multiple USB ports and USB cables with my computer :-/

But so far the USB connection seems to be the error, even if I can ping the device and the network is up and running; no services are listed or ports scan-able.

Do you have any idea where it "could" be possible to deactivate all services or connections via USB on the device itself? I mostly tried out all settings I could find in the UI, but so far I'm out of spots to look at.

Is it possible to run `nmap` on the internal USB IP when the USB connection is active? Do you see some ports then on your machine or nothing?

xsacha, 2016-09-13:
You can try hardworking 192.168.0.1 in the app to see if it is working properly and just not being detected.

prasadeng, 2017-03-10:
i also have same error in linux mint please  resolved it
"Communication Error: 1 (Connection refused) from 192.168.1.3"

klango, 2017-11-18:
Hi I ended up here when had the problem connecting my Passport (10.3.3.1435)

1. Sachesi 2.0.3
2. Dev mode on
3. Mass storage off
4. Errors
 "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" and
"Communication Error: 203 (Error downloading http://192.168.2.105/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.2.105"

Tried all suggested procedures with no luck. I changed the IP Adress on BB10 under Development mode to 169.254.0.5 (instead default 169.254.0.1)
Sachesi then was able to recognize. I hope this can help someone.

Cheers

gstanden, 2017-11-18:
Hey I'm just so moved that there are people still using the amazing BlackBerry OS10 devices, and, hats off to Sacha for his continued support of the BlackBerry OS10 community. Thanks Sacha!                                                                                                                                                                         Sent from my BlackBerry - the most secure mobile device                                                                                                                                                                           From: notifications@github.comSent: November 18, 2017 11:54To: Sachesi@noreply.github.comReply-to: reply@reply.github.comCc: subscribed@noreply.github.comSubject: Re: [xsacha/Sachesi] (Linux) Communication Error, even with udev rule (#93)  Hi I ended up here when had the problem connecting my Passport (10.3.3.1435)
Sachesi 2.0.3Dev mode onMass storage offErrors
"Communication Error: 4 (Socket operation timed out) from 169.254.0.1" and
"Communication Error: 203 (Error downloading http://192.168.2.105/cgi-bin/discovery.cgi  - server replied: Not Found) from 192.168.2.105"
Tried all suggested procedures with no luck. I changed the IP Adress on BB10 under Development mode to 169.254.0.5 (instead default 169.254.0.1)
Sachesi then was able to recognize. I hope this can help someone.
Cheers

xsacha, 2017-11-18:
Thanks.
I just wish blackberry still supported it and my device didn't die.

klango, 2017-11-19:
Just  a little update on this, I figured it out the connection problem.

While changing the IP Adress on BB10 under Development mode OR RIGHT AFTER connecting to the pc, you need to assign dhcpclient (dhcpclient <interface> ) in my case enp0** 

@xsacha just curiosity, what device(s) have you been using lately?

xsacha, 2017-11-19:
Huawei P9 Plus was device I bought after my Blackberry. Still have it