xsacha/Sachesi #95: Sachesi 2.0.3 uses incorrect IP for my Z10

pbierhoff, 2016-10-27:
I have just compiled Sachesi 2.0.3 for my 32-bit Ubuntu 16.0.4.1 LTS install. (BTW: I had to add 'sudo apt-get install libudev-dev' as an extra requirement, and had to explicitly specify qmake to use QT5 by issuing 'qmake -qt=5 j4;' )

Trying to connect to my Z10 - I have a proper IP for my device:

```
enp0s29f7u6 Link encap:Ethernet  HWaddr <snip>  
          inet addr:169.254.241.94  Bcast:169.254.241.95  Mask:255.255.255.252
          inet6 addr: fe80::6ecd:bb21:5997:99ec/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:83 errors:59 dropped:0 overruns:0 frame:59
          TX packets:288 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:30210 (30.2 KB)  TX bytes:52415 (52.4 KB)

```

When running Sachesi, it sees my device as a Windows device (correctly), tries to connect, but spews out these messages:

```
"Communication Error: 4 (Socket operation timed out) from 169.254.241.93"
"Communication Error: 4 (Socket operation timed out) from 169.254.241.93"
"Communication Error: 4 (Socket operation timed out) from 169.254.241.93"
"Communication Error: 4 (Socket operation timed out) from 169.254.241.93"
"Communication Error: 4 (Socket operation timed out) from 169.254.241.93"

```

The IP address mentioned in the messages is 1 off from the Z10. (On second thought, this might be the IP of my computer). I've tried adding usb0 to /etc/network/interfaces but no dice. How do I get a working connection?
My Z10 is still running the T-Mobile provided 10.2.1 - was hoping to sideload a newer version as T-Mobile won't provide it.

xsacha, 2016-10-27:
This is the correct IP address. The one printed (94) is your computer. The one it is trying to connect to is the phone (93).

Do you know if the phones IP is not 93? I don't know a way to find it and so I subtract 1 from host IP.

pbierhoff, 2016-10-27:
Yeah, you are right. .94 is my computer, so .93 is the presumed address of my phone. I have not found a way to get the IP address on the phone - 10.3.1 seems to have a way to get it but not 10.2.
Any suggestions as to what I can try to get the connection working? I'm going to try a reboot next.

pbierhoff, 2016-10-27:
Reboot didn't help. Phone connects fine, shows up in ifconfig. Sachesi just keeps spinning. I assume the password box is the passcode to unlock my Z10, correct?

pbierhoff, 2016-10-27:
I can ping my Z10:

$ ping -b 169.254.120.179
WARNING: pinging broadcast address
PING 169.254.120.179 (169.254.120.179) 56(84) bytes of data.
64 bytes from 169.254.120.177: icmp_seq=1 ttl=255 time=1.83 ms
64 bytes from 169.254.120.177: icmp_seq=2 ttl=255 time=2.86 ms
64 bytes from 169.254.120.177: icmp_seq=3 ttl=255 time=1.86 ms
64 bytes from 169.254.120.177: icmp_seq=4 ttl=255 time=2.14 ms
64 bytes from 169.254.120.177: icmp_seq=5 ttl=255 time=2.18 ms
64 bytes from 169.254.120.177: icmp_seq=6 ttl=255 time=2.39 ms
^C
--- 169.254.120.179 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5006ms
rtt min/avg/max/mdev = 1.833/2.215/2.865/0.352 ms

xsacha, 2016-10-27:
It is the password for developer mode, yes.
Is developer mode turned on?

pbierhoff, 2016-10-28:
First of all, thank you SO much for the fast responses. Much appreciated.

I do NOT have Developer Mode turned on. I was convinced that it should not be, but if you tell me otherwise, I'll turn it on.

I'm not sure what the difference is between the password to unlock my phone, and the password to enable Developer Mode. Do I set the latter when I enable that mode?

xsacha, 2016-10-28:
If all you are doing is installing third-party apps then you technically would not require developer mode. Most of Sachesi features do require it (installing firmware, system apps, technical info, bootloader). So the login assumes this method.

pbierhoff, 2016-10-30:
Enabled Developer Mode. Tried again. Same story. Detects device, cannot talk to it.So I switched from 32bit Linux to 64bit Windows 10. Sachesi 2.0.3 ran flawlessly, upgraded my Z10 to 10.3.2 (from 10.2.1) using the Rogers Canada software update. No issues whatsoever. Issue might still exist in the 32bit Linux compile I did, but my personal issue is now closed.

closed 2016-10-30 by pbierhoff

xsacha, 2016-10-30:
I have to admit I never tried 32-bit Linux so it is possible something in there causes issues (like type sizes).
For Linux x86, pretty much everyone uses 64-bit.