xsacha/Sachesi #96: Windows release should be packaged as regular .zip file

kbroulik, 2017-01-02:
Windows can handle .zip files out of the box, it cannot cope with .7z, however, and would require me installing a 3rd party decompression tool for no reason.

pierredGitHub, 2017-05-15:
There are several different types of ZIP files (different levels of compression). Windows can create only the basic type, but can read more than the basic type. The advantage of the .7z extension is that it is highly compressed. 

That being said, if I am not mistaken, 7ZIP can create zip files that are still highly compressed but can still be decompressed by Windows. It may not be as compressed as .7z but still.

I would tend to agree that a .zip distribution for Windows would make more sense. I am saying this even though I have 7ZIP installed on my computer.

xsacha, 2018-05-12:
Ideally the builds should all be static as the project is opensource and can comply with licence restrictions.
In that situation, there would not be issues with using zip.

xsacha, 2018-05-20:
2.0.4 now has a zip option

closed 2018-05-20 by xsacha