xsacha/Sachesi #97: debian compilation

mencacci, 2017-01-02:
Hello there I can't compile with openssl 1.1.0, I get 
In file included from /usr/include/openssl/asn1.h:24:0,
                 from /usr/include/openssl/rsa.h:16,
                 from src/installer_establish.cpp:17:
src/installer_establish.cpp: In member function ‘void InstallNet::requestChallenge()’:
src/installer_establish.cpp:37:72: error: invalid use of incomplete type ‘RSA {aka struct rsa_st}’
     unsigned char* privkey_mod = new unsigned char[BN_num_bytes(privkey->n)];
                                                                        ^
In file included from /usr/include/openssl/crypto.h:31:0,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/asn1.h:16,
                 from /usr/include/openssl/rsa.h:16,
                 from src/installer_establish.cpp:17:
/usr/include/openssl/ossl_typ.h:110:16: note: forward declaration of ‘RSA {aka struct rsa_st}’
 typedef struct rsa_st RSA;
                ^~~~~~
src/installer_establish.cpp:38:22: error: invalid use of incomplete type ‘RSA {aka struct rsa_st}’
     BN_bn2bin(privkey->n, privkey_mod);
                      ^~
In file included from /usr/include/openssl/crypto.h:31:0,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/asn1.h:16,
                 from /usr/include/openssl/rsa.h:16,
                 from src/installer_establish.cpp:17:
/usr/include/openssl/ossl_typ.h:110:16: note: forward declaration of ‘RSA {aka struct rsa_st}’
 typedef struct rsa_st RSA;
                ^~~~~~
Makefile:943: recipe for target 'installer_establish.o' failed
make: *** [installer_establish.o] Error 1
instead with your compiled linux version I get 
/usr/lib/x86_64-linux-gnu/libcrypto.so.1.0.0: version `OPENSSL_1.0.0' not found (required by ../Sachesi)

Any hint? thanx

xsacha, 2017-01-02:
For some reason it compiled with libcrypto 1.0 and is looking for that.
I will have a look later.

xsacha, 2017-05-13:
Yeah you will need an older OpenSSL as 1.1+ has a new API.

xsacha, 2018-05-12:
As 1.1.x has a newer API, you will need to modify the source code to comply (a few #if OPENSSL_VERSION > 101000 for eg.).

Otherwise, OpenSSL 1.0 still works and every major distro should include a copy.

Are you saying debian does not have it available in repository?

closed 2018-05-12 by xsacha