xsacha/Sachesi #100: Need concise compilation instructions for Ubuntu 16.x

tstrike34, 2017-02-07:
Various packages are being substituted in this version of Ubuntu. Using the Ubuntu 14.x compilation instructions are outdated for Ubuntu 16.x.

xsacha, 2017-05-13:
I am unsure of the specific package names but essentially you just need Qt5 and libusb.

JDJoe, 2017-05-13:
you need to remove qtdeclarative5-controls-plugin, qtdeclarative5-dialogs-plugin qtdeclarative5-quicklayouts-plugin and run:

$ sudo apt-get install qt5-qmake libqt5network5 libqt5qml5 qtdeclarative5-settings-plugin qtdeclarative5-window-plugin qtdeclarative5-qtquick2-plugin qttools5-dev-tools libssl-dev qtdeclarative5-dev libusb-1.0-0-dev

make sure you have libudev-dev installed too

xsacha, 2017-05-13:
Thanks @JDJoe

closed 2017-05-13 by xsacha