xsacha/Sachesi #99: Fedora 25

tstrike34, 2017-02-07:
Having communications problems via USB with Sachesi 2.0.2 and a compiled 2.0.3 (from source code).

Is there a way to have an extended debug log somewhere?

[00:24:06] Communication Error: 4 (Socket operation timed out) from 169.254.202.69
[00:24:06] Communication Error: 4 (Socket operation timed out) from 169.254.202.69
[00:24:06] Communication Error: 4 (Socket operation timed out) from 169.254.202.69
[00:24:06] Communication Error: 4 (Socket operation timed out) from 169.254.202.69
[00:24:06] Communication Error: 4 (Socket operation timed out) from 169.254.202.69

Fedora 25 Core (201) - 4.9.7-201.fc25.x86_64

xsacha, 2017-02-08:
Thats all the information you will ever get. The USB device wont talk to
you. Probably your OS is blocking it. Do you need permission to access it?
This is Linux?

tstrike34, 2017-02-08:
Yes it is Linux.                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone on the Verizon Wireless 4G LTE network.                                                                                                                                                                                                                From: SachaSent: Tuesday, February 7, 2017 7:03 PMTo: xsacha/SachesiReply To: xsacha/SachesiCc: tstrike34; AuthorSubject: Re: [xsacha/Sachesi] Fedora 25 (#99)Thats all the information you will ever get. The USB device wont talk to
you. Probably your OS is blocking it. Do you need permission to access it?
This is Linux?

camellan, 2017-11-12:
I solved the connection problems, installing packages qml-module-qt-websockets and qml-module-qtwebsockets.

xsacha, 2018-05-12:
External issue

closed 2018-05-12 by xsacha