xsacha/Sachesi #101: 2.0.3 built on linux 64-bit drops connection after a couple of minutes

rolfep, 2017-02-22:
Howdy,
I am running Sachesi on ROSA Desktop Fresh R8 release.  When I start Sachesi and connect my Q10 running 10.3.3.2205, a second possible device is detected and, soon, Sachesi loads all the info of my Q10.

I go to the backup tab, choose all three checkboxes, choose to back up blind, and select a save file location and name.

The spinner spins, progress stays at 0%, and, after a couple of minutes, the connection wizard is running, showing only one possibility, which I think is my HDHomeRun device on the lan.

The connection log contains:  

**The following message is repeated until the connection to Q10 is made:**

08:58:52] Communication Error: 203 (Error transferring http://192.168.1.115/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.1.115
[08:58:52] Connected to MSM8960 V3.2.1.1 F NEWPORT Rev:17 at 169.254.0.1.
[08:59:22] Communication Error: 203 (Error transferring http://192.168.1.115/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.1.115
[08:59:22] Connected to MSM8960 V3.2.1.1 F NEWPORT Rev:17 at 169.254.0.1.
[09:04:39] Communication Error: 4 (Socket operation timed out) from 169.254.0.1
[09:04:40] Communication Error: 203 (Error transferring http://192.168.1.115/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.1.115

**The previous message is repeated until I quit `Sachesi.`**

192.168.1.115 is the static dhcp address I assign to the HDHomeRun in OpenWRT on my router.
192.168.1.218 is assigned to the Q10, btw, and I have that set by static dhcp in order to bookmark ftp connections and so forth.

My LAN is on 192.168.1.xxx and the usb-connected Q10 is reported by ifconfig at 169.254.0.2

I thought devices had to be on the same network to 'talk'?  If I go into the Q10 usb settings to try to set an ip address on my LAN, the choices are limited to only start with 169.

Thanks.

rolfep, 2017-02-22:
On another machine on the lan, running Mageia release 5 (Official) for x86_64, I built 2.0.3 but get an error when trying to start it:

```
$ Sachesi 
Cannot mix incompatible Qt library (version 0x50201) with this library (version 0x50402)
Aborted
```
I can run Sachesi-1.4.0, here, ifconfig shows the ip as 169.254.0.2, and I just completed a backup to an NFS share on the LAN:

```
$ ll /mnt/stuff/q10/sachesi/20170222-full.bbb 
-rw-rw-r-- 1 rolf rolf 3667206547 02-22-2017 09:56 /mnt/stuff/q10/sachesi/20170222-full.bbb
```

xsacha, 2017-02-22:
Unfortunately it is trying to connect to the HD Home Lan as it appears in the list and somehow met the criteria and had the right port open. I haven't run in to that issue before.
I guess the app needs to be changed to ignore that. In some versions it looks only for 169.*. You could temporarily disable wifi as a workaround.

rolfep, 2017-02-25:
All the devices on the LAN are wired, except phones, and turning off wifi on the Q10 does not change the behavior, is probably not what you meant. 

I pulled the ethernet cord from the [HDHomeRun TV tuner](https://www.silicondust.com/product/hdhomerun-extend/) and started Sachesi 2.0.3 with Q10 connected.  It was recognized and, as before, I go to the backup tab, choose all three checkboxes, choose to back up blind, and select a save file location and name.  After a few minutes of spinner and 0% progress reported, the connection is lost.  The log is simpler, no HDHomeRun ip, only two lines:

```
[05:02:58] Connected to MSM8960 V3.2.1.1 F NEWPORT Rev:17 at 169.254.82.61.
[05:05:13] Communication Error: 4 (Socket operation timed out) from 169.254.82.61
```

I've been trying a lot of different strategies of downloading, building, and running Sachesi so, for a little precision, this Sachesi executable came from a download of Sachesi2.0.3-Linux.tar.bz2

Thanks.

camellan, 2017-11-12:
I solved the connection problems, installing packages qml-module-qt-websockets and qml-module-qtwebsockets.

klango, 2017-11-18:
Hi I ended up here when had the problem connecting my Passport (10.3.3.1435)

1. Sachesi 2.0.3
2. Dev mode on
3. Mass storage off
4. Errors
 "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" and
"Communication Error: 203 (Error downloading http://192.168.2.105/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.2.105"

Tried all suggested procedures with no luck. I changed the IP Adress on BB10 under Development mode to 169.254.0.5 (instead default 169.254.0.1)
Sachesi then was able to recognize. I hope this can help someone.

Cheers

Vistaus, 2018-01-23:
I have the same problem. Tried @klango's suggestion of changing the IP address to 169.254.0.5 but that doesn't help :( Running Solus (Linux).

klango, 2018-01-23:
Try dhcpclient and the adapter name

Vistaus, 2018-01-23:
@klango Thanks, but that doesn't help either :(

camellan, 2018-01-31:
`camellan@ladoga:~$ cat /etc/network/if-up.d/blackberry_dhcp
#!/bin/sh

if [ "$IFACE" = enp0s19f2u3 ]; then
        dhcpcd enp0s19f2u3
fi

if [ "$IFACE" = usb0 ]; then
        dhcpcd usb0
fi
camellan@ladoga:~$ 
`

xsacha, 2018-01-31:
Thanks. This is a common issue on particular distros that do not automatically connect USB network devices.
Perhaps for security reasons.

Vistaus, 2018-01-31:
@camellan @xsacha That's nice and all, but on Solus (which I'm using) there is no /etc/network folder.

xsacha, 2018-01-31:
Do you have ability to run a dhcp daemon on an interface in Solus?

Vistaus, 2018-01-31:
Yes, I already tried that several times to no avail. I mean: the command runs fine and stuff, but Sachesi still doesn't pick up my Passport.

camellan, 2018-02-01:
In my case, dhclient did not work either, and dhcpcd did everything perfectly. And try installing packages qml-module-qt-websockets and qml-module-qtwebsockets.

Vistaus, 2018-02-01:
@camellan Thanks. That did not really help much, although one particular error in the terminal is gone now. But Sachesi still doesn't connect.

Here's the output from dhcpcd:
[I] ╭─vistaus@solus /home/vistaus/Downloads  ‹system› 
╰─$  sudo dhcpcd enp0s20u2                                                                                                                                                                                                   ↵ 1
DUID 00:01:00:01:22:05:ca:41:a6:e4:b8:36:4b:a9
enp0s20u2: IAID b8:36:4b:a9
enp0s20u2: soliciting an IPv6 router
enp0s20u2: soliciting a DHCP lease
enp0s20u2: offered 169.254.0.2 from 169.254.0.1
enp0s20u2: leased 169.254.0.2 for 604800 seconds
enp0s20u2: adding route to 169.254.0.0/30
enp0s20u2: adding host route to 169.254.0.1
forked to background, child pid 1784
[I] ╭─vistaus@solus /home/vistaus/Downloads  ‹system› 
╰─$ sudo dhcpcd usb0
usb0: interface not found or invalid
dhcpcd exited
[I] ╭─vistaus@solus /home/vistaus/Downloads 

So it seems to invoke the first command just right, but the usb0 is not a valid interface?

camellan, 2018-02-01:
This means that the usb0 interface you have. The name of the network interface Blackberry - enp0s20u2. You set the suggested me in the previous post?

Vistaus, 2018-02-01:
Yes.

camellan, 2018-02-03:
Display the output of the "ifconfig" command.

klango, 2018-02-03:
In my case dhcpclient works and dhcpcd doesnt lol. Very akward. Running
opensuse here.

https://ci6.googleusercontent.com/proxy/AhxK5b-f4HzJ8829rygfBc44fzmUpBwNeF9uJbs8ciO8WkH00yPwVtoJsiaIY8sYsRvIOSkqlWVSRyNoV5l17szKfMZe38IGhASWHPYEk7nRtmxM27FZWFexJ1UrjtdiWnZ_SuK0-FWtqaUsdDLoR0L1Up22Zw=s0-d-e1-ft#https://github.com/notifications/beacon/ALF0TiBXZz-DFxG0dbB_Wb3nPyBh0Q_Xks5tRJUBgaJpZM4MI8aG.gif>

Vistaus, 2018-02-05:
@camellan 

```
╭─vistaus@solus /home/vistaus/Downloads  ‹system› 
╰─$ ifconfig
enp0s20u2 Link encap:Ethernet  HWaddr A6:E4:B8:36:4B:A9
          inet addr:169.254.205.130  Bcast:169.254.205.131  Mask:255.255.255.252
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:47 errors:0 dropped:0 overruns:0 frame:0
          TX packets:96 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:11022  TX bytes:14710

enp0s20u2i2Link encap:Ethernet  HWaddr A6:E4:B8:36:4B:AB
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0  TX bytes:0

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Bcast:0.0.0.0  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1431 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1431 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:112575  TX bytes:112575

wlp1s0    Link encap:Ethernet  HWaddr 58:91:CF:D3:0B:B2
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:672647 errors:0 dropped:0 overruns:0 frame:0
          TX packets:519133 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:665389603  TX bytes:167493765

[I] ╭─vistaus@solus /home/vistaus/Downloads  ‹system› 
╰─$ 
```

camellan, 2018-02-06:
Well, the network is all right. Try installing the libssl0.9.8 package in addition. There are no more ideas ...

Vistaus, 2018-02-07:
@camellan There's no libssl0.9.8 package in Solus. Where can I download the source code of libssl 0.9.8?

xsacha, 2018-02-07:
If you compile from source you can use whatever libssl you want.
The binaries have a static libssl too. So not sure how this can be a problem.

Vistaus, 2018-02-09:
I finally got it to compile from source again and now it connected just fine! Thanks for all of you guys help! :)

xsacha, 2018-02-09:
Awesome news

klango, 2018-02-09:
nice to know! great support from you guys

closed 2018-05-12 by xsacha