xsacha/Sachesi #102: Sachesi has stopped working

pierredGitHub, 2017-05-12:
Hi, I am using a BB Z10 running software release 10.3.3.1463, OS version 10.3.3.2205. 3-4 months ago I wiped my phone to fix an issue with backups (after removing an enterprise account), and reinstalled everything from scratch. I recovered the ability to do backups using BBLink. Fine. Then about 6 weeks ago, backups starting failing again with "Backup could not be completed .... Make sure your device is connected ....".

Searching online, I found that Sachesi could solve that problem. So I tried a backup with it (Sachesi 2.0.3 on Windows 10). First, I can't get the size info (timeout ...., supposedly a BB bug). But waiting a few more seconds and it populates after I clicked OK on the error message. But that doesn't happen every time. More often that not the sizes do show up. Weird, but anyway.

So I do a backup. At 9%, I get "Sachesi has stopped working. A problem cause the program to stop working correctly. Windows will close the program and notify you if a solution is available.", and all I can do is click "Close program". That window is titled "Sachesi" and doesn't look like what I am used to. Seems Sachesi specific rather than Windows generic. 

One interesting bit of info. When it is going to fail, the percent count on the BB continues to climb, while the count on the PC does not. That's how I can tell it is about to fail. The count on the BB continues for another 8-9%. Not sure if that gives any valuable info.

So I suspect recently installed apps. So I remove a few of them that are new (not the updated ones) but that only changes when Sachesi's backups fail. I have had backups issues in the past (with BBLink) but at least it was hanging in the middle of the backup so I could see what it was backing up at the time of the hang. After removing the offending app, backup started working again. But I can't do that with Sachesi because it doesn't show what it is backing up while doing it.

In the Event Viewer, I see an Error at the time of failure with the following info:
Faulting application name: sachesi.EXE, version: 2.0.3.0, time stamp: 0x567585fc
Faulting module name: ucrtbase.dll, version: 10.0.14393.0, time stamp: 0x57898db2
Exception code: 0xc0000409
Fault offset: 0x000891cb
Faulting process id: 0x1a54
Faulting application start time: 0x01d2cb2a910dcb26
Faulting application path: E:\Installation\Sachesi2.0.3-Windows\sachesi.EXE
Faulting module path: C:\WINDOWS\System32\ucrtbase.dll
Report Id: 503a569c-ad12-4f6c-ba98-938fd4e3f2a4
Faulting package full name: 
Faulting package-relative application ID: 

Is there a way to get a log from Sachesi? Or any other way to figure out why Sachesi fails?

Thanks for your help.

xsacha, 2017-05-13:
The sizes not showing up is a Blackberry bug and very normal for large backups, which is why I recommend backing up 'blind' (not checking size).

There are other Blackberry bugs with backing up like the app-specific issue you discovered and the 4GB limit for each type.

I suppose I could show each app that is currently being backed up and that would resolve your issue?

xsacha, 2017-05-13:
Ah, the problem is I do not know what the apps are without first checking size (which won't work and is probably why you are using Sachesi im

closed 2017-05-13 by xsacha

pierredGitHub, 2017-05-13:
Well, as I said in my original post, the sizes do show a few seconds after I click OK most of the time. So it might still be worth trying to display the app name while backing up.

Another tidbit of info, I tried it on another Win10 computer that had a newer ucrtbase.dll because I installed the Creators Update. On that computer the backup seems to complete, but not every time. The reason I say "seems" is because the one time I was there to watch the end, it was at 70% (settings 100%) and then the progress information simply disappeared. It didn't say the backup was complete or anything else.

I tried it again and the 2nd backup is exactly the same size (but I didn't see the end). Trying it a 3rd time and it failed (I didn't see) as the backup is half the size.

Any information you could add to the display would be helpful. If you display a percentage (as you are currently doing) then you must know how big it is going to be, right?

Thanks for your help.

xsacha, 2017-05-13:
The device sends me a percentage only during backup AFAIR.
I only know size if i query it (which doesn't work for many people).
If querying size works, Blackberry Link should also work for you.

pierredGitHub, 2017-05-13:
I understand what you are saying. But Link is extremely buggy (more so on Win10, for which it has never been updated). At least Sachesi is still being updated, thanks to you. If I attempt a backup, it fails while computing backup data (after about a minute or so), saying "Backup did not complete. Verify that your device is turned on and connected to BlackBerry Link, and try again".

The message is so bogus it's not even funny. I wouldn't be able to start a backup if the device wasn't connected. And on top of that, I can do a Sync Calendar/Contacts so Link is obviously communicating with the device.

This got way worse when I upgraded my computer from Win7 to Win10. So I suspect a timing issue aggravated by Win10. I have to find a computer that still has Win7, just to check that theory. In the mean time Sachesi is my only savings grace, in the long run.

As I said, on Win10 Creators Update, Sachesi seems to have completed a backup. The size is a bit larger than the last successful backup with Link, which makes me think it's ok, but Sachesi was advertising a 7.6GB backup and the file is only 5.1GB. 

Does Sachesi say something at the end when the backup is successful? Or does the percent display simply disappear?

Thanks for putting up with me on this.

xsacha, 2017-05-13:
To be fair, the problem isn't Link itself but its assumptions. When querying the size, the device attempts to create an actual zipped and encrypted backup of all your files and then report its size to you. When this times out (due to http timeout on your OS), it is reported as no reply and disconnection so the device can never report that size. The main failure is no progress updates.

If you are able to update the timeout somehow it would work better in Link but I have not inbestigated this too far. My attempts in Windows did not change the timeout. The device itself does not timeout.

I think if the backup is over 4GB, it will never succeed in Sachesi due to a limitation in my zipping third-party library (Quazip). It tries to write but writes to the wrong offsets. The backup would have been successfully read off the devoce, hence the success message.

Sachesi reports success when percent becomes 100 AFAIR.

pierredGitHub, 2017-05-13:
Oh oh, so you are saying that my backups cannot work with Sachesi, and never will, since they are over 4GB (currently 5GB)? That's too bad. If so, then would you consider switching to a newer library that supports file larger than 4GB?

I made a 5GB backup yesterday with Sachesi (still not sure whether it was complete). Are you saying that garbage is most likely garbage?

xsacha, 2017-05-13:
If it told you it is 7GB, I would expect a 7GB file.
If it said success and gave you a 5GB file, in my mind, that would be the quazip bug.
 I have tried to track down the >4GB bug before but couldn't find the issue. This was 2 years ago I think so QuaZip may be improved now (or not). It could certainly be worthwhile switching to something else.

xsacha, 2017-05-13:
Did you check whether the backup is valid? Can Link see it?

pierredGitHub, 2017-05-13:
I just tried. As soon as I click on Restore, and therefore before I can even select which backup to restore, I get "BlackBerry Link has stopped working". Tried it 3 times. This is scary. I really need a Win7 computer. I am building a VMware image of Win7 right now to see if that will help (I don't have a free computer to wipe and install 7 on right now. They are all on Win10 already. So disappointing that BB never saw fit to upgrade BBLink for Win10.

pierredGitHub, 2017-05-15:
Hi Sacaha,

As I write this, I am doing a backup of my Z10 using BBLink in a virtual machine running Win7. The same Z10 and BBLink that would fail in Win10. So it is obviously something in Win10 that broke Link. 

The backup started with a temp file (.dtmp) of 7.3GB, size consistent with what Sachesi was announcing. The size of that file is continuously decreasing in size as the backup progresses. So there must be some compression going on. The reason I am saying this is that the 5.4GB backup that Sachesi created a couple of days ago might be fine. I will have to test if I can open it from Link in Win7.

Will keep you posted. It feels good to see a backup going :-)

xsacha, 2017-05-15:
Ok. Thanks for the update.

pierredGitHub, 2017-05-15:
Update: The backup completed. Trying to restore anything still resulted in Link "has stopped working", even in the Win7 virtual machine. At first I thought it was because I was accessing the backup through a share. That was not it as copying them locally didn't help. So I renamed the extension of the two recent backups, the one done with Sachesi and the one just done with Link in the VM. Link no longer crashed. Renamed the Sachesi backup back and it crashed again. So that was the reason. Probably the >4GB restriction or something. So deleted the Sachesi backup. Back to using shares, I can read the latest backup or the older ones from 2 months ago from the VM image, even when the backup files are on a share.

Back into Win10, Link no longer crashes when trying to restore (since the Sachesi backup is gone), but it still spins indefinitely while "Loading the restore content...".

So Win7 in a VM is the only reliable solution for me. It's apparently a timing thing since Link in Win7 is so fast at detecting and connecting to the device, whereas it's hit and miss in Win10, it can take 4-5 tries before being successful at connecting. And it takes several minutes when successful (vs seconds in Win7).

Last, I tried running Link in Win7 compatibility mode under Win10. The restore is still spinning on "Loading the restore content ..." after 10 minutes.

So, looks like I can rest easy again knowing I can make backups of my Z10. But that solution (a virtual machine) is not for everyone. I hope someone can benefit from this exchange of the past couple of days. I think you should still look into updating the Sachesi library. Thanks.