xsacha/Sachesi #103: Cant connect any more - Detected 1 Blackberry USB device(s) in Autodetect Mode

JDJoe, 2017-05-13:
Sam problem, as https://github.com/xsacha/Sachesi/issues/60 

I use Kubuntu Linux 16.10 64-bit 
Qt 5.6.1
Kernel 4.8.0-51-generic 

Phone is Q10
Software release 10.3.3.164
OS version 10.3.3.2205

Phone gets the IP via USB

$ifconfig 
```

    enp0s20f0u5: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.0.2  netmask 255.255.255.252  broadcast 169.254.0.3
        ...
```

and both versions of Sachesi (Sachesi2.0.2 and 3)  are spitting out:
```

    "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" 
    "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" 
```

or what ever IP phone has -1

I enabled and disabled dev mode on my phone - no change 
My PC can see the phone obviously and and phone gets a IP address

$ tcpdump -i enp0s20f0u5

```
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on enp0s20f0u5, link-type EN10MB (Ethernet), capture size 262144 bytes
15:08:18.881545 IP PC7.47676 > BLACKBERRY-84AD.local.http: Flags [S], seq 1201939367, win 29200, options [mss 1460,sackOK,TS val 401512 ecr 0,nop,wscale 7], length 0... 

```

I am out of ideas and need your help.

xsacha, 2017-05-13:
Doesn't look related to #60 as #60 was in mass storage mode. Yours looks to be correctly in RNDIS/ethernet mode but isn't allowing connection on the port.

I do not have a physical device anymore so I am not sure if the newer firmwares changed anything. Not sure how else to help sorry!

JDJoe, 2017-05-13:
Is there any debugging info I can post here?

JDJoe, 2017-05-13:
dev mode off:

```
enp0s20f0u5: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.103.98  netmask 255.255.255.252  broadcast 169.254.103.99

```
and 
```

qrc:/qml/generic/Installer.qml:90:25: Unable to assign [undefined] to QUrl
mart@PC7:~/Downloads/Sachesi2.0.3-Linux$ 
mart@PC7:~/Downloads/Sachesi2.0.3-Linux$ 
mart@PC7:~/Downloads/Sachesi2.0.3-Linux$ ./Sachesi 
qrc:/qml/generic/Installer.qml:90:25: Unable to assign [undefined] to QUrl
"Communication Error: 4 (Socket operation timed out) from 169.254.103.97" 
"Communication Error: 4 (Socket operation timed out) from 169.254.103.97" 

```

same time I get:

tcpdump -i enp0s20f0u5

> tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
> listening on enp0s20f0u5, link-type EN10MB (Ethernet), capture size 262144 bytes
> 15:49:42.177545 IP PC7.47406 > BLACKBERRY-84AD.local.http: Flags [S], seq 4014659891, win 29200, options [mss 1460,sackOK,TS val 1022336 ecr 0,nop,wscale 7], length 0
> 15:49:44.358869 IP PC7.48594 > BLACKBERRY-84AD.local.http: Flags [S], seq 1943284196, win 29200, options [mss 1460,sackOK,TS val 1022881 ecr 0,nop,wscale 7], length 0
> 15:49:45.249550 IP PC7.47612 > BLACKBERRY-84AD.local.http: Flags [S], seq 2493479781, win 29200, options [mss 1460,sackOK,TS val 1023104 ecr 0,nop,wscale 7], length 0
>...

xsacha, 2017-05-13:
That looks like perfectly normal output to me. The unable to assign is a known one which has no effect.
The IP looks like the right one.
I wouldn't know how else to work through this issue without debugging the app om your hardware.
Can you compile the app at all?

JDJoe, 2017-05-13:
>  Can you compile the app at all? 

No, I can not. It fails with: 

```
....
-lQt5Gui -lQt5Qml -lQt5Network -lQt5Core -lGL -lpthread 
/usr/bin/ld: cannot find -ludev
collect2: error: ld returned 1 exit status
Makefile:295: recipe for target 'Sachesi' failed
make: *** [Sachesi] Error 1
```

xsacha, 2017-05-13:
It is the same version so it should behave the same but you should be able to set a breakpoint on the request to IP, verify it is correct and a breakpoint on the receive callback and verify it doesn't occur.

Outside of that it would mean the device doesn't run the normal web server anymore.
You could force it to skip the checks.

JDJoe, 2017-05-13:
>It is the same version so it should behave the same but you should be able to set a breakpoint on the request to IP, verify it is correct and a breakpoint on the receive callback and verify it doesn't occur.

>Outside of that it would mean the device doesn't run the normal web server anymore.
>You could force it to skip the checks.

Umm... can you elaborate on this because I have no idea how to do any of those things. :)

xsacha, 2017-05-13:
Sorry,
I don't know how I can help remotely but you are hitting this code: https://github.com/xsacha/Sachesi/blob/master/src/installer.cpp#L703

And you can try to force it to think it is connected and hope it works by changing it (within the braces {}) to these lines:
        setIp(ip_addr);
        setState(1);
        checkLogin();

JDJoe, 2017-05-13:
https://github.com/xsacha/Sachesi/blob/master/src/installer.cpp#L717 
718 719 

looks now like this: 
        // Valid device
        setIp(ip_addr);
        setState(1);
        checkLogin();

recompiled and it has no effect.

xsacha, 2017-05-13:
OK, then it might be a problem with the newer firmware changing the login l/web server or the windows/linux 'mode' the RNDIS is set to. Sorry I can't help further.

JDJoe, 2017-05-13:
OK, thanks. Lets hope someone else can :)

JDJoe, 2017-05-13:
As it turned out, the problem is not Sachesi. The problem is the **network manager**.
Workaround is here: https://github.com/xsacha/Sachesi/issues/69#issuecomment-124849325 

Looks like we can add this to the list of Kubuntu network-manager problems.

closed 2017-05-13 by JDJoe

xsacha, 2017-05-13:
Interesting. I actually used KDE as my primary desktop when writing this app and hadn't experienced this.
Maybe new issue or just kubuntu?

Thanks for the problem solving!

mbirth, 2017-11-06:
I have this in Ubuntu 17.10. It worked fine in 17.04. I already tried disabling the connection in NetworkManager, but that doesn't change anything. The network adapter gets the correct ip (169.254.0.2) and I can ping 169.254.0.1.

However, when starting Sachesi, all I get is:

```
$ ./Sachesi
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model"
qrc:/qml/generic/Installer.qml:90:25: Unable to assign [undefined] to QUrl
```

And there's the progress pie chart in a window "Install" with the word "Sending" and "0%" that doesn't go away. (I can close that, though.)

Sachesi's main window reads "Detected 1 Blackberry USB device(s) in Autodetect mode." and spins indefinitely. No output on the command line.

xsacha, 2017-11-06:
Could be network manager related like previous poster?

mbirth, 2017-11-06:
Nah, tried disabling the connection there and running dhclient manually. Didn't change a thing. Also from a networking perspective, the connection is there.

klango, 2017-11-18:
Hi I ended up here when had the problem connecting my Passport (10.3.3.1435)

1. Sachesi 2.0.3
2. Dev mode on
3. Mass storage off
4. Errors
 "Communication Error: 4 (Socket operation timed out) from 169.254.0.1" and
"Communication Error: 203 (Error downloading http://192.168.2.105/cgi-bin/discovery.cgi - server replied: Not Found) from 192.168.2.105"

Tried all suggested procedures with no luck. I changed the IP Adress on BB10 under Development mode to 169.254.0.5 (instead default 169.254.0.1)
Sachesi then was able to recognize. I hope this can help someone.

Cheers

klango, 2017-11-19:
Just a little update on this, I figured it out the connection problem.

While changing the IP Adress on BB10 under Development mode OR RIGHT AFTER connecting to the pc, you need to assign dhcpclient (dhcpclient ) in my case enp0**

mbirth, 2018-05-20:
I just had time to experiment again. I've noticed the 169.254.0.1 IP seems to only be reachable via WiFi. Using only the cable, I get the 169.254.0.2 IP on the interface on my PC, but the .1 isn't accessible. But as soon as I enable WiFi on the BB Classic, I can ping .1 just fine (via my WiFi router, it seems). However, Sachesi doesn't see the device.

This only happened after the 10.3.3.2206 update. Sachesi worked fine before that.

xsacha, 2018-05-21:
@mbirth Thanks for the update.
Was the compiled binary able to work or you needed to compile yourself?

mbirth, 2018-05-21:
I had to compile it myself - against Qt 5.9. There's no Qt 5.10 for Ubuntu at the moment, so the precompiled one from the GitHub download didn't work.

xsacha, 2018-05-21:
Ah right. Thought 5.10 was old enough since 5.11 comes out tomorrow.
The OpenSSL update was the only major change to 2.04.

mbirth, 2018-05-21:
Must be something with OpenSSL or some other issue with Ubuntu then. So for anyone else having connectivity issues with Ubuntu: Download the source code, install the required packages as per the README and compile Sachesi yourself.