xsacha/Sachesi #112: build issue on

L1cardo, 2019-06-10:


L1cardo, 2019-06-10:
I don't know how to fix it.
Can you build a Mac version please！

xsacha, 2019-06-10:
You need to have openssl development headers. Install openssl off the official website to get these.

L1cardo, 2019-06-10:
I don’t know too much about coding, so it will be difficult for me to fix the issue.
OpenSSL doesn’t have a compiled version for Mac, i don’t know what to do and I have installed OpenSSL through homebrew, what should I do next?

xsacha, 2019-06-11:
Try these instructions:
https://medium.com/@timmykko/using-openssl-library-with-macos-sierra-7807cfd47892

L1cardo, 2019-06-11:
I am done with it ! One after another !
There is too many problem with the 2.0.3 version ,so I wanted to build a 2.0.4 version . but now I quit .
I don't know much about coding or compiling, I am just a normal user. it is not easy for me to solve the problem.

[image]

xsacha, 2019-06-11:
That's the same problem. It can't find your OpenSSL library. You need to tell it where it is if it isn't in a standard location.
You use the -L flag in the same way you used the -I flag.

L1cardo, 2019-06-12:
I don't know how but finally I have built it. but there is no any translation !

[image]

xsacha, 2019-06-12:
It should have automatic translations based on your locale.
Which language?

L1cardo, 2019-06-13:
> It should have automatic translations based on your locale.
> Which language?

It should be Chinese. But it is English and there is no language switch button.

xsacha, 2019-06-13:
It's meant to work on your systems language. There's no switch for it. It's a Qt thing.

L1cardo, 2019-06-13:
> It's meant to work on your systems language. There's no switch for it. It's a Qt thing.

My system language is Chinese.

L1cardo, 2019-06-14:
@xsacha Can you build a macOS version for me please, I will be grateful!
Thanks!

xsacha, 2019-06-14:
I don't think it'd be any different to the one you have.

I'd have to install macos or run it in a VM (which is how I made the other versions) and then set it up.

L1cardo, 2019-06-15:
For now I have two issue , I am wondering if there was something wrong with my build 
1.The version 2.0.3 that you built works on macOS10.15 only a few crashes, but version 2.0.4 can not work on macOS10.15.
2.The version 2.0.3 that you built has a language switch button , but 2.0.4 doesn't .

[image]

[image]

xsacha, 2019-06-15:
Oh that is strange. I don't remember ever seeing this language switch.. and I compiled it!
Possibly you need to run a special qmake command to generate the languages?

closed 2019-11-03 by L1cardo