xsacha/Sachesi #117: Not Working

Ziegan, 2021-03-04:
After launching the Sachesi it detects the BlackBerry device and after that i encounter this problem.

and the connection get failed.

_qrc:/qml/generic/USBConnect.qml:46:13: QML Button: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
QIODevice::write (QTemporaryFile, ""): device not open
qrc:/qml/generic/Installer.qml:90:17: Unable to assign [undefined] to QUrl
qrc:/qml/generic/USBConnect.qml:46:13: QML Button: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead._

xsacha, 2021-03-04:
Is that related to permission to write to a temporary file? Does it have write permission in the folder?

Ziegan, 2021-03-05:
Sorry, for the delayed response.
I'll check it ASAP and let you know if its related to permission issue !

Ziegan, 2021-03-05:
Sorry for the problem to be posted without any further thinking about it.
Its really a problem related to permission and now its working flawlessly.

Closing this issue.

Thank you for your contribution to blackberry, loving this tool !

closed 2021-03-05 by Ziegan