xsacha/Sachesi #118: Add rescue functions

pwrsurge, 2022-03-01:
I currently have a Passport which keeps going in read only mode about 2 minutes after boot as per https://forums.crackberry.com/blackberry-10-os-f269/bb10-rescue-mode-1193787/  The phone works fine during these 2 minutes and after that, no new apps can be opened.  However, existing apps still continue to function but can't save any data to the device memory as it is read only.

Sachesi is able connect and successfully reinstall BAR files within these 2 minutes but the connection times out after that and any transfer still in progress gets aborted and the connection to the device is lost.  I was able successfully reinstall individual system BAR files including the radio but nothing has worked so far.  However, I haven't been able to reinstall the factory BAR file as the transfer times out before it has a chance to install due to the large file size.

As I have apps which don't have any BAR files avaiable that are used on a daily basis, I need to fix this issue without wiping the phone.

By viewing the source, I see there is code to put the device in boot loader mode when wiping the device but would like to know if there are additional functions available in boot loader mode which could help me.  Perhaps a button could be added to put the phone in boot  loader mode without doing anything else which would be useful if this still allows BAR files to be installed on the device.  Also, adding support to execute recovery utilities such as chkqnx6fs or to open a recovery shell would be great.

xsacha, 2022-03-01:
Sorry, I don't have any other special functions that I didn't expose.
The bar files you want may exist on the internet?

xsacha, 2022-03-07:
Bootloader mode doesn't allow transferring files. You have to upload signed RAM images instead.
Sachesi can restore backups.

pwrsurge, 2022-03-08:
As I always have about 2 minutes after booting the phone before it becomes read only, is there any way to implement a function to pause it from fully booting?  After the boot screen shows 25%, I can start transferring BAR Files but the phone becomes read only before I have the chance to reinstall all of them or even only the system factory BAR file as it is over 300MB in size.

Also, could you please implement a function to view any sort of boot or application logs so that I can try to figure out exactly what is causing my phone to crash and become read only.

Even since the BBID server was shutdown, Sachesi shows an error when going to the Backups tab.

xsacha, 2022-03-09:
If it requires BBID server, that'd be on the device side and I can't remedy that sorry :(

The boot logs require a physical JTAG connection unless you have access to the filesystem.

You could wipe it via bootloader but this won't help you get these .bar files back or activate the BBID. Not sure there's anything else that could be done.

pwrsurge, 2022-03-09:
I don't think it requires BBID server, as long as the phone being restored to has a matching BBID.  I have another BlackBerry Passport which already has the same BBID on it so I want to try backing up this phone and restoring it on that one.

I believe you had added some extra validation check with the BBID server in one of your previous versions which needs to be disabled now that the server is gone.

Could you please look into this to see if anything can be done?  At least, we need to get a working solution to continue being able to backup and restore devices.

xsacha, 2022-03-09:
As far as I knew, it didn't even require internet for backup/restore. I certainly never check any server so can only assume that is being done on-device. The backup method changed from Playbook to BB10 to capture more data. I'm not sure the old method still works.

You can test older versions if you believe the older method (action request) works on BB10 but I fear it will skip app backups anyway.

There's a special user for backup/restore. It's the same one that handles other web requests such as device info.