xsacha/Sachesi #119: Playbook Bootrom, MLO, Bootloader

avktech78, 2022-06-14:
Hello, could you please share the bootrom, mlo, etc.?
https://forums.crackberry.com/blackberry-playbook-os-f228/playbook-bootrom-mlo-bootloader-830909/

xsacha, 2022-06-14:
It has been posted before and was taken down. I had it on mega (now deleted) and Google Drive. Will take me a while to find it again.

avktech78, 2022-06-14:
Thanks for the quick response.
I would like to try to get root or learn how to change the firmware. The interest is more of an academic one.
About 13 years ago I was researching blackberry v5-7, but destroyed all the work. Now I have some free time and some BB10 phones and want to go back to research.
If you will have a chance to look for these files, it would be great. If not, that's okay.
And thank you for Sachesi, it is very helpful in research.

xsacha, 2022-06-14:
@alexhude should be able to help you there if he's free.

closed 2022-06-14 by xsacha